ondrej1024/ddguard #4: Medtronic 640G

nickxbs, 2020-04-26:
Hi, is it also possible to read from the 640G?

ondrej1024, 2020-04-26:
I think it should work as long as you are using a Contour Next Link 2.4 USB stick to communicate with the pump. The communication stack is based on the implementation from the project [decoding-contour-next-link](https://github.com/pazaan/decoding-contour-next-link) and if the 640G is supported by this then also DD-Guard supports it. 
I have only my daughters 670G to test so cannot confirm this. Why don't you give it try and report back if it works for you ?
Thanks

nickxbs, 2020-04-26:
Cool! Yes, i will try.
Thanks

nickxbs, 2020-04-27:
Yes, it works! The update is every 5 minute. Just a small detail... why when nightscout is updated it shows that the data is from 5 minute ago?

ondrej1024, 2020-04-27:
Great. Thanks for the feedback. So the 640G is officially supported :smile: 

As for the time reported in Nightscout, this is something I need to fix. Basically when DD-Guard starts it starts polling the pump status every 5mins and doesn't adjust the polling period. So, in the worst case, the first sensor reading could already be 4:59 mins old when DD-Guard starts just a short time before the pump receives the next sensor reading. The sensor reading is reported to NS with the actual time of the reading as reported by the pump (so 4:59min ago). So in the worst case you get the sensor readings 5min late.

I need to modify DD-Guard to poll the pump status shortly after the next sensor reading is expected. This is on my TODO list.

nickxbs, 2020-04-27:
Ok, make sense!
I'm not a python expert so i cant help you :) but let me know if you need some test.

My setup is Medtronic 640G(CNL2.4), Raspberry Pi 3 and Nightscout 13.0.1

closed 2020-04-30 by ondrej1024