ondrej1024/ddguard #7: Would it be possible to use the cnl code on an esp32 with micropython

chachonen, 2020-08-05:
Hi, I was wondering if the code (only talking about the contour next module from pazaan) could be adapted and deployed to an esp32 using micropython? The idea would be to then use BLE advertising to broadcast the data to an app on a mobile device or as you, an rpi that may/may not forward to other deices for remote monitoring. I have no experience with python/micropython and would just like to know your opinion about this.

Thanks
Henrik

ondrej1024, 2020-08-05:
The idea sounds interesting. However, I have no experience with Micropython. I guess it has to be checked if the packages we are using for the CNL driver are available for Micropython. 
Is it possible to connect the CNL stick to the ESP32 ?

chachonen, 2020-08-06:
Ha, ha... Sadly, It does not seem that easy. So it's back to the original plan with rpi zero. Do you have any suggestion (mostly regarding powering the rpi) to make it as wearable as possible, the idea was to make it as small and light as possible as it should be worn along side the pump. 

Thanks

ondrej1024, 2020-08-06:
Right, there is no such thing as "easy" in electronics. :smile: 

I am using a RPi-Zero which is powered with this [battery pack](https://www.amazon.it/dp/B075455DRN). It powers the whole system for about 10h and you can charge it without switching off the RPi. See picture below of the assembled sytem (CLN, RPi and battery pack). As you can see, the biggest part is the CNL stick, so I don't think you can make it any smaller. Anyway, it fits nicely in a small backpack.

[image]

chachonen, 2020-08-06:
Seems the ESP32-S2 has OTG USB. (https://www.electronics-lab.com/meet-iot-ready-esp32-based-maia-development-board). Plus I've seen somewhere the usb been dismantled (only if you don't mind loosing the blood sugar measuring functionality). No idea of electronics, but it might be an option. To me making it the smallest possible for a kid is THE THING (they already have to run around with a couple of devices on them).

Anyway, thanks for the good quick feedback and best regards!

ondrej1024, 2020-08-06:
Actually I don't really understand how your system would work and why you want to use the ESP32 and BLE. How would this amke the system any smaller?
If you want to receive the pump data via CNL directly on a smartphone you can simply use the [AndroidUploader](https://github.com/pazaan/600SeriesAndroidUploader) app.

chachonen, 2020-08-06:
I know... I don't express myself clearly... The goal for me would be to have a small device that is able to extract the latest data from the pump and forward it. BLE would be my choice as it does not require to part of any network (it could just act as a beacon sending out data to any device listening). The observing devices (in my case a mobile phone) could be given to a teacher (a person near, for BT 4.2 I think it's a matter of meters, but for 5 and up it is increased considerably). This would allow for constant monitoring and the observing device (which obviously needs to be relatively near) can upload the data to the cloud if needed (nightscout...). It would also allow the CNL to be near the pump and constantly be connected and send data out. Later in "future" versions it might be possible to create BLE mesh networks and include the device in it/them. Specially intended for small kids and for the tranquility of the caring people. That's it. Whatever the device I actually do not care, as long as it is small, light weight and does the job. I just mentioned the ESP32 as it is able to run micropython and the  code communication with the CNL is written in python. I appreciate any opinion/suggestion and thank you for code!

Regards
Henrik

ondrej1024, 2020-08-06:
The radio link between Minimed 670G pump and CNL stick has similar range than BLE, maybe even more. So if you put a gateway device like the one in the picture above in the classroom (or give it to the teacher) it would receive the pump readings and the child does not have to wear any additional device (other than the pump and the sensor). The gateway can be connected to the cloud via Wifi.

Also a phone with the Android Uploader and the CNL connected given to the teacher would do the same job.

Looks like with your idea you are just adding another device between the CNL and the gateway.

coss30, 2020-10-30:
scusate se ho ben capito il ddguard è un'alternativa al nightscout non riesce a fare boli da remoto o per lo meno non rientra nei suoi obbiettivi? scusate l'ignoranza....ringrazio in anticipo

ondrej1024, 2020-10-30:
Remote bolusing is not supported by DD-Guard. It can be used only for monitoring.