ondrej1024/ddguard #9: "Next reading" broken when starting ddguard without sensor connection

linus-ha, 2021-01-02:
When starting ddguard at the time when no sensor is connected, the received values for "sensorBGL:" and "sensorBGLTimestamp" are 0, so that the next reading will be scheduled for the 01.01.1970 at 01:05 which will never be reached.

```
[...]
### BGL ###
sensorBGL:                0
sensorBGLTimestamp:       1970-01-01 01:00:00+01:00
trendArrow:               -3

account for pump RTC drift:
   before: pumpTime 2021-01-02 21:29:47.999971+01:00,  sensorBGLTimestamp 1970-01-01 01:00:00+01:00
   after : pumpTime 2021-01-02 21:28:22.652260+01:00,  sensorBGLTimestamp 1970-01-01 01:00:00+01:00
Uploading data to Nightscout
Sensor lost, not uploading SGV data
Next reading at 1970-01-01 01:05:00+01:00, -1609619005 seconds from now
```

linus-ha, 2021-01-02:
I've encountered  a second bug which is fixed as well, when testing for negative "seconds from now"-values (time in the past).
````
[...]
### BGL ###
sensorBGL:                197
sensorBGLTimestamp:       2021-01-02 21:35:37.999996+01:00
trendArrow:               0

account for pump RTC drift:
   before: pumpTime 2021-01-02 21:40:46.999977+01:00,  sensorBGLTimestamp 2021-01-02 21:35:37.999997+01:00
   after : pumpTime 2021-01-02 21:39:21.198504+01:00,  sensorBGLTimestamp 2021-01-02 21:34:12.198524+01:00
Uploading data to Nightscout
Next reading at 2021-01-02 21:39:12.198524+01:00, -11 seconds from now
````

The time is calculated by the wrong time, so that the "next reading" can be scheduled for a time in the past.

linus-ha, 2021-01-02:
Okay sorry, it's just the log that's wrong. The scheduling works alright and doesn't behave like what the printed values are showing, which means, that the log is incorrect. Not that big of a problem.

ondrej1024, 2021-01-03:
Hi @linus-ha ,

can you please check if this commit 8e0ef86e78b70bf70e428bd8b241a345740392ad solves the issue?