ondrej1024/ddguard #11: Transmitter battery drain

hognefossland, 2021-04-09:
Hi Ondrej1024,

Thanks for your excellent work on this application. I've set it up on an Raspberry Pi 3 to monitor my daughter's blood sugar levels. Her pump is a 640G.

I had been using the Android 600 series uploader, but wanted a solution that would be permanently powered, and your solution is perfect for that.

However, since starting to use DD-Guard, the Guardian 3 transmitter battery no longer appears to be lasting the full 7 days. Instead it runs out after only 3-6 days.

Due to Covid, my daughter has been spending a lot of time at home and is being monitored pretty much 24/7, whereas previously we'd use the Android solution only during the night (for  nearly three years), so this might be a contributing factor.

It's only been a few weeks at this point, but I'm curious whether you've ever noticed any battery impact from using DD-Guard. Of so, do you have any suggestions as to what night help mitigate the issue? Presumably reducing the polling rate will improve the situation.

I did come across a thread on the 600 series uploader that also seemed to suggest there might be an impact on the batteries from using these tools, though I'm struggling to understand how the transmitter battery (rather than the pump battery) would be affected.

Any insight would be much appreciated.

Thanks,
Hogne

ondrej1024, 2021-04-10:
Hi Hogne,

I have not seen any correlation with using DD-Guard and transmitter battery drain. I don't know how these 2 things would be connected. I think the transmitter sends a new reading to the pump every 5 mins. This should not change, no matter how  often you request the pump status. In the beginning I was afraid the _pump_ battery would discharge faster with DD-Guard requesting the status every 5 minutes but I tested this with and without DD-Guard and the pump battery always lasted about 2 weeks.

The transmitter battery here lasted barely 6/7 days, although the sensor mostly died before this, so we almost never got to see the battery discharge completely. One thing which is important though is using a good quality battery in the charger. My daughter used the 670G, but has now switched to the 780G with the GL3 transmitter. This one is still at about 40% after 7 days using the Medtronic Minimed app.

The other discussion you mentioned is [this](https://github.com/pazaan/600SeriesAndroidUploader/issues/270) one I guess. There was also no correlation found between using Nightscout and battery drain but there are many other things which can influence this.

I guess the only thing you can do is not using DD-Guard for some days and check if the transmitter battery discharges slower.

Cheers, Ondrej

hognefossland, 2021-04-10:
Hi Ondrej,

Thanks for your response.

Totally agree with you - what you're describing is exactly my understanding too, i.e. that the _pump_ battery would potentially be impacted (though I can't say I've seen any signs of that yet), but not the _transmitter_ battery. Thought I'd ask anyway.

Interesting point on the charger battery -- that's not something I had actually considered at all. I'm not quite sure how that would impact on things either, but I'll keep that in mind going forward.

As it happens, we've only yesterday started a new transmitter (even though the previous one was only about 6 months old), with a new charger and a fresh (Medtronic-supplied) battery. I'll keep an eye on things for a few weeks, and try with and without DD-Guard and see if I can uncover anything interesting.

How's the 780G and the phone app that you mentioned? Do you know if several phones can be connected at the same time?

Thanks,
Hogne

ondrej1024, 2021-04-11:
Once I tried to use a rechargable battery in the transmitter charger. That was not a good idea because the transmitter charged this way lasted only about 4 days. I think it is a questions of the charger battery voltage level. If you use a partially discharged battery or rechargable battery the voltage level is lower than a fresh quality battery. And this leads to the transmitter not being charged completely, so it will discharge more quickly.

The 780G communicates with the Smartphone of the T1D where the "Minimed Mobile" app needs to be installed. This is the uploader which receives the pump data via BLE and then sends it to the CareLink Cloud service. From there, up to 5 care givers can receive the data with the "Carelink Connect" app. It provides basically all the data which can be downloaded from the pump. Also alarms and history data for 1 day are provided. 

At the moment there are  no alternative ways of downloading the pump data from the 780G, which is not a big issue as the official apps work quite well. However, it is a challenge which might be worth tackling :smirk:

hognefossland, 2021-04-13:
Hi,

I've never tried a rechargeable battery in the charger, but those are interesting observations. Perhaps making sure to change the battery more frequently will make a difference.

The 780G sounds like a big jump from the 640G/670G. It's just sad that it has taken this long to get there. It's not like communicating directly with a phone is revolutionary in this day and age. Only question is, how can we get our hands on one? I don't know what it's like in your country, but here (in Ireland) they're provided through the public system, which probably means we're less likely to get access to the latest versions straight away. Your answers have definitely convinced me we need to push for an upgrade though.

Thanks again,
Hogne

ondrej1024, 2021-04-16:
Also here in Italy we are provided with the insulin pump by the public health service. 
But the device they offer you also depends on the region where you live and the local policies. We were lucky to be offered to switch from 670G to 780G.

As you can see from the screenshot below (taken from the Carelink Connect app), with less than a day of sensor life left, there is still a 56% of transmitter battery charge.

[image]