ondrej1024/ddguard #12: ddguard program

frapala, 2021-06-27:
Hi,

I tried to run the ddguard.py program on ubuntu os but it send me some error like `configparser.NosectionError: No section: 'blynk'` on lines 159, 781 and 1149 and other error `'ConfigParser' has no attribute 'NoOptionError'` on lines 401 and 162. Whereas helpers.py, cnl24driverlib.py... work. I install all the library so I don't understand why it don't run well.

I am a french student and I work on a research project on insulin patch and I wanted to know (if I understand well to work it needs wifi), why can't we connect it with bluetooth to work whenever we want even without connection like if someone wwant to go in a trip without wifi.

Cordially.

ondrej1024, 2021-06-28:
> I tried to run the ddguard.py program on ubuntu os but it send me some error like `configparser.NosectionError: No section: 'blynk'` on lines 159, 781 and 1149 and other error `'ConfigParser' has no [...]

That seems actually to be a bug. You can work around this by replacing all lines
`
except ConfigParser.NoOptionError as NoSectionError:
`
with
`
except:
`

> I am a french student and I work on a research project on insulin patch and I wanted to know (if I understand well to work it needs wifi), why can't we connect it with bluetooth to work whenever we want even without connection like if someone wwant to go in a trip without wifi.

If I understand you correctly, by looking at the [DD-Guard architecture](https://github.com/ondrej1024/ddguard#how-it-works), you would like to skip the Cloud connection and have a direct Bluetooth connection from the gateway to the phone. This is a good idea and would be indeed very useful if you are traveling and have no Wifi. But it's just not implemented. You would need to add Bluetooth communication to the gateway and implement an app which receives the data on the phone via Bluetooth.

Cheers, Ondrej

frapala, 2021-06-28:
Thank you for the help. I will try that. But before I have a problem I don't know why when I do `systemctl start ddguard` to launch the program I reiceve `Failed to start ddguard.service: Unit ddguard.service not found` but I don't know which program it launch because all the other .py work well.
Cheers, Frapala

ondrej1024, 2021-06-29:
Did you run the `install.sh` script? If yes, did it throw any errors?