ondrej1024/ddguard #13: Full pump history to NS

sztypa, 2022-04-01:
Hi,

Any chance to get a full pump history uploaded to NS?
Simply just like [600SeriesAndroidUploader] (https://github.com/pazaan/600SeriesAndroidUploader) does?

BTW, nice work.

I'm running it on RPi 3B+ and Intel NUC Linux Minta based, but missing "all this funny stuff" like basal rates, carbs, etc from my 670G.

sztypa

p.s. Is project still alive or it just died because you're looping now with something else (or got 780G now?)

ondrej1024, 2022-04-02:
Hi,

the implementation of pump history handling is not working reliably in the cnl24driverlib. But this was fixed in the improved driver by @oldsterIL in [his repository](https://github.com/oldsterIL/600SeriesDriver).

So starting from there you would have to implement the uploading of this data to NightScout.

I cannot help with that as my daughter has switched her 670G with the 780G pump. so I just don't have the hardware anymore.

And now I am focusing on development of a tool for the 780G: [M5Stack Minimed Monitor](https://github.com/ondrej1024/m5-minimed-monitor)

Cheers, Ondrej

sztypa, 2022-04-04:
Thank You.
I'm interested in projects like your's begause of my job. I'm working as a nurse with diabetic patients, teach in use of insulin pumps and have been DM1 sinde 1992 (pump user since 2009: 522->640g->Veo 554->670G->...)

oldsterIL, 2022-04-05:
You can use my [project](https://github.com/oldsterIL/600SeriesDriver), it outputs all the history data from the pump. History can be of two types: sensor history, pump history. Unfortunately, there is not much information left that I could not make out, but there is very little of it.

If you need help feel free to contact me, I'm ready to help.

Regards, Denis.

sztypa, 2022-04-20:
Thank You for reply. As Your project doesn't support data upload to NS, i may try to find out with android device and Link 2.4 connected via OTG cable.
It's working 80% of time now with RPi3 with Linegage OS (Android) installed on it, but it's not perfect due to power failure issiues aka E86 error).

oldsterIL, 2022-04-22:
As far as I know, error E86 occurs if the device (RPI, mobile phone, etc.) "can not get in time" to read data from the pump. Especially often this occurs when reading a "history" from a pump. When reading "history" from the pump, many data packets are transmitted. If the device (RPI, mobile phone, etc.) does not have time to read them, error E86 occurs.

Try using a Android phone and 600SeriesAndroidUploader.

sztypa, 2022-04-27:
Thanks. I'll do.

Or give it a try with Intel NUC based Android.